chipKIT32/chipKIT32-MAX #549: Have to use INPUT to set an output

Berryfier, 2015-02-06:
e.g. TRISBbits.TRISB15 = INPUT sets RB15 for output
Using mpide 0023 with chipkit pi

majenkotech, 2015-02-06:
Why would you think that was an error? You're mixing chipKIT defines with native PIC32 register manipulation.  Of course it doesn't work.  Use 1 for input or 0 for output as specified by the PIC32 documentation, or use pinMode(pin, INPUT) or pinMode(pin, OUTPUT). Don't mix the two.

EmbeddedMan, 2015-02-06:
Berryfier,

Matt is correct - using the TRIS register (and the INPUT and OUTPUT
constants) is a "bare metal PIC32" thing, and the Arduino environment
already defines the constants INPUT and OUTPUT, and unfortunately those
definitions are not compatible with the ones that the TRIS bits need.

One way around this would be to define your own TRIS_INPUT and TRIS_OUTPUT
constants, and use those when setting up the TRIS registers.

*Brian

majenkotech, 2015-02-06:
Or just use numbers and remember it's `1`nput and `0`utput.

Berryfier, 2015-02-06:
Thank you for the explanations and describing the proper use.  
I was following the examples numbered 1 and 1.1 here:
http://wiki.pinguino.cc/index.php/Pic32-pinguino-MX220
It's all helpful and I've learned a lot.

majenkotech, 2015-02-06:
Pinguino is a completely different system.  Their code is only applicable to using their software on their boards.

closed 2015-02-06 by Berryfier