chipKIT32/chipKIT32-MAX #551: chipKIT boards show up as Teensy boards in Windows Device Manager

JacobChrist, 2015-02-13:
When the Teensy drivers are installed chipKIT stk500v2 drivers show up as Teensy:

[image]

EmbeddedMan, 2015-02-13:
I have tried to reproduce this, and have been unable to. I installed Arduino 1.0.6, then Teensyduino (which installed it's USB drivers). I then took a Teensy 2.0, compiled a sketch for it that used USB serial, downloaded it and confirmed it's operation. While plugged in, the Teensy shows up as "Teensy USB Serial (COM4)" as expected.

I then unplugged the Teensy, and plugged in a Fubarino Mini with a serial based sketch. The Fubarino shows up as chipKIT USB Serial (Stk500v2 compatible) (COM14). 

With both plugged in at the same time, I get proper USB names for both boards.

This is under Windows 8.1 64 bit.

JacobChrist, 2015-02-14:
We see this on two different Windows 7 computers.

EmbeddedMan, 2015-08-09:
I feel like this is some strange driver mis-match issue under Windows. I can't reproduce the issue under Windows 7 or Windows 8, but then my development system is probably pretty 'unique' and doesn't represent a typical user's machine.

To help me debug this problem could you plug in both a chipKIT USB board (like Fubarino) and a Teensy board and then run USBView.exe (http://www.softpedia.com/get/System/System-Info/USBView.shtml) and send me a screenshot with those two fully expanded?

ricklon, 2015-08-21:
I've seen this problem as well. I've had to reset Windows drivers. This was on Windows 7 machines.

JacobChrist, 2015-08-21:
Could this have anything to do with chipKIT using a VID/PID assigned by
FTDI?

EmbeddedMan, 2015-08-21:
Great question. I'm not sure what the mechanism would be though. We use an
FTDI VID of 0x0403  with a custom (hopefully unique) PID of 0xA662. If
Teensy is also using the same VID and PID as us, then I think they're
probably doing something wrong. :-) But my guess is that they are not using
those VIDs and PIDs.

My guess is that there is an error in either their INF file or ours that is
stomping on the other INF file. Remember, we don't have our own driver - we
just have an INF file that says to Windows "when you see this VID/PID
combo, use your built-in usbser.sys driver". Both their INF and our INF
file say that same thing.

This issue needs more investigation. I'm happy to do it, but it's going to
be a while before I can find the time.

*Brian

JacobChrist, 2015-08-27:
I have new information. I just plugged in an Arduino Uno and it shows up as a Teensy in the Device Manager.  Teensy is just rude.

EmbeddedMan, 2015-08-27:
There must be some INI magic that Paul's doing in the Teensy driver install
INF file that allows this behavior. I'm not a wizard with INF files and
don't know how, but that's what must be going on.

*Brian