chipKIT32/chipKIT32-MAX #553: MPIDE 0023-macosx-20140821 — fatal error: xc.h: No such file or directory#446

rei-vilo, 2015-02-14:
MPIDE 0023-macosx-20140821 fails to compile the blink example and reports the following error
   
> 
/Applications/IDE/Mpide-20140821.app/Contents/Resources/Java/hardware/pic32/cores/pic32/cpp-startup.S:45:16: fatal error: xc.h: No such file or directory
compilation terminated.
>

However, xc.h is located at Mpide-20140821.app/Contents/Resources/Java/hardware/pic32/compiler/pic32-tools/pic32mx/include/xc.h

[image]

Same as #446 but for a more recent release.

EmbeddedMan, 2015-02-14:
I've seen similar sorts of things before. All you need to do is just recompile again, and it should work. It only happens for me very rarely, and normally just the first time I use a new version. Try that and see if it works for you.

rei-vilo, 2015-02-14:
Unfortunately, recompiling doesn't solve the problem.

EmbeddedMan, 2015-02-14:
OK, shoot. I'll try to reproduce the problem here.

*Brian

rei-vilo, 2015-02-14:
I performed a fresh build from the repository.

* `Mpide` runs fine only when located in the `/Applications` folder. It fails in any other folder, e.g. in `/Applications/IDE`.
* Similarly, `Mpide` runs fine only with its genuine name. It fails for any change of the name, e.g. `Mpide150`.

This is usually caused by some hard coding of files paths and names in the code.

ricklon, 2015-02-14:
Some how a hard coded path must be in that version or the current version
of the code.  I've had issues with Macs that have a case sensitive file
system.

--Rick

ricklon, 2015-02-14:
Here's the most likely reason:
https://github.com/chipKIT32/chipKIT32-MAX/blob/master/app/src/processing/app/debug/Compiler.java#L142

```				basePath += "/mpide.app/Contents/Resources/Java";
```

rei-vilo, 2015-02-14:
Good catch @ricklon !

ricklon, 2015-03-12:
This issue occurs when compiling through Java instead of a makefile. The makefile skips this Java code.

rei-vilo, 2015-03-12:
@ricklon 

Conversely as in #494, is there something Java does but a makefile misses when compiling?

ricklon, 2015-03-12:
I think the issue for #494 is in platforms.txt. The new flags aren't being used in the makefile.