chipKIT32/chipKIT32-MAX #561: Inconsistent boards.txt chipKIT Files

rei-vilo, 2015-04-09:
All the `boards.txt` files for the chipKIT mention `compiler.c.flags` and `compiler.cpp.flags`, except WiFire, WF32 and Max32.

```
uno_pic32.compiler.c.flags=-O2::-c::-mno-smart-io::-w::-ffunction-sections::-fdata-sections::-G1024::-g::-mdebugger::-Wcast-align::-fno-short-double
uno_pic32.compiler.cpp.flags=-O2::-c::-mno-smart-io::-w::-fno-exceptions::-ffunction-sections::-fdata-sections::-G1024::-g::-mdebugger::-Wcast-align::-fno-short-double
```

Please add those two missing entries to the 3 boards.

ricklon, 2015-04-10:
Thanks for finding that. Can do.

rei-vilo, 2015-04-10:
You're welcome! I've included a patch on my embedXcode makefiles to cope with this.

ricklon, 2015-04-11:
I'll check with Keith about that value, but it should.

KeithV, 2015-06-15:
The compiler options come from platforms.txt and do not need to be overridden in boards.txt, so this is not an error. The flash size is an error and is 2^^21 - 2^^14 (MZ have 16K pages, not 4K pages). The actual flash size is 2080768.

rei-vilo, 2015-06-16:
So, in that case, wipe the `compiler.c.flags` and `compiler.cpp.flags`out of the `boards.txt` for better clarity.

ricklon, 2015-06-16:
Good idea. If it's not actively maintained or there for a reason it should
go away because it will hide fixes or changes that happen in platforms.txt.

EmbeddedMan, 2015-10-21:
OK, at this point there are default compiler.c.flags and compiler.cpp.flags in the platforms.txt file, and then there are over-ridden values in boards.txt for any board that can benefit from the -G1024 compiler option. (which turns out to be most of them). At this point I feel like this issue has been taken as far as it needs to go.