chipKIT32/chipKIT32-MAX #564: .NET Framework 3.5 install required for Windows 8 install

JacobChrist, 2015-05-23:
When installing chipKIT drivers (chipKITDriverInstaller_v10.exe) in Windows 8 on a MS Surface Pro 3 tablet. The following installation of .NET Framework 3.5 is required.  Not a show stopper, but inconvenient and consumes additional space on small SSD table drives.

[image]

majenkotech, 2015-05-23:
Is it the installer for the drivers that needs .NET or the driver itself? My guess is the installer.  Can we re-package with something lighter?

JacobChrist, 2015-05-23:
I don't know.  Installing the driver + .NET only took up about 1M of HDD space, so not a huge issue.  There are bigger fish to fry, but I just wanted to log the issue, since most new computers will be Win 8+.

Before:
[image]

After:
[image]

EmbeddedMan, 2015-08-09:
Yes, Matt, you are correct. It is only the installer that needs .NET, not the drivers themselves. The driver are simply an INF file. The reason we have the .exe (which uses .NET) is because many folks have a hard time installing bare .INF files. So we use a driver installer that I created for the Egg-Bot project (where people have the same problem) based on the Nullsoft Install System. It is the piece that requires .NET.

Jacob, I think the chipKITDriverInstaller.exe is still a good thing to have, and I don't feel like it's worth our time to re-write it so that it doesn't need .NET. What would be a good way to educate users that, if they don't want to download .NET, they can install the drivers manually? To me that's a better solution.

JacobChrist, 2015-08-09:
Don't need to rewrite but maybe having Win7 and Win8 versions would take
care of it.

majenkotech, 2015-08-09:
Have you come across Nullsoft scriptable installer (NSIS) ? It's what I use to package UECIDE.

http://nsis.sourceforge.net/Main_Page

Available for Windows, linux, you name it.  Generates a .exe on virtually any platform and doesn't need (AFAIK) .NET to be installed for the installer to run.

EmbeddedMan, 2015-08-09:
Matt,

Hmm. Maybe I'm mixing something up then -  I used Nullsoft installer to
generate the installer in question. I just assumed it needed .NET, but if
it runs on the other platforms, clearly it doesn't. Hmmm. More
investigation is necessary.

Jacob,

I'm not quite sure how creating a Win7 and Win8 version of the installer
would help. _IF_ the installer is the one requiring .NET 3.5, then any OS
that doesn't have it will need it to be installed first. I'm not sure what
I'd do differently when creating the separate installers for Win7 and Win8
that would make a difference . . .

*Brian

majenkotech, 2015-08-09:
Maybe you are using an older version? I don't recall ever needing to have .NET installed to run the generated installer, but then .NET is something I normally have installed anyway...

EmbeddedMan, 2015-08-09:
Well, I think I made the installer several years ago at least. So it was
likely an earlier version. I do always have .NET installed, so I wouldn't
ever know if it needed to be installed. Hmm. I wonder how I can test this.
I don't want to buy a Windows licence just to throw it on a VM so I can
create a clean starting place for each test. But maybe that's the best way
to do it.

*Brian

JacobChrist, 2015-08-10:
We install the driver on a new Win8 surface pro 3 about once per week.  I
think we usually have about 5 laying around.  Let me check which versions
of .NET come on a new surface.

Jacob

EmbeddedMan, 2015-08-11:
OK, I think I understand things a bit better now. The Nullsoft installer does NOT need .NET, it's the actual executable (USBDriverInstaller.exe) that is unpacked and run by Nulloft that needs .NET. That application is written in C# with MSStudio 2008 express, and so will require at least a certain version (or higher) of .NET to be installed in order to run. One way to get around this would be to see if the Nullsoft installer could actually do the driver install for us, rather than having a separate .exe (C#) to do that piece of the install. I'll look into that.

majenkotech, 2015-08-11:
It can do it for us in anything below Vista, but in Vista and upwards we just need Nullsoft to execute the command `pnputil.exe /a /i <path to extracted inf file>` (I think) as part of its post-extraction routines.

EmbeddedMan, 2015-08-11:
Good call Matt - this page
https://technet.microsoft.com/en-us/library/Ff800798.aspx indicates that
pnputil.exe is available on all systems Vista and beyond. So this looks
like the perfect way to not have to use the separate .exe file so as to not
require .NET.

The only folks that would be left out would be WinXP people - is that a big
deal?

*Brian

majenkotech, 2015-08-11:
I'm sure we can find out what OS they are on within Nullsoft and use the InstDrv plugin to Nullsoft for those few numpties.  It would be nice if that plugin worked with pnputil but it looks like it doesn't.