chipKIT32/chipKIT32-MAX #572: SoftPWMServo library crashes when you pass it non-existent pin

EmbeddedMan, 2015-06-19:
For example, on a FubarinoSD, which has 45 total I/O pins, if you do
SoftPWMServoServoWrite(45, 1500);
the library will crash. This should not happen. Even if you pass in a non-existent pin, it would be nice if the library didn't crash the sketch.

majenkotech, 2015-06-19:
I see there is no protection at all in any of those functions - you should check for pin < NUM_DIGITAL_PINS in all functions that are passed a pin.

EmbeddedMan, 2015-06-19:
Matt,

Agreed. I would like to know why it crashes though - maybe there's a more
subtle bug that can be fixed, as well as limit checking the pin numbers.

*Brian

majenkotech, 2015-06-19:
Firstly there is a hard coded limit of 86 pins in the library (SOFTPWMSERVO_MAX_PINS).  This seems a bit redundant since there is already a board specific number of digital pins, which should be the maximum pins the library needs to deal with.

Also arrays are being defined with SOFTPWMSERVO_MAX_PINS entries, then checking if pin numbers are *greater* than that, so it's possible to access an invalid array slice as pin 86.

Not sure what the crash with lower pin numbers is though...

EmbeddedMan, 2015-06-19:
Matt,

Yup, good point. I'll change that and update the library.

*Brian

closed 2015-06-19 by EmbeddedMan