chipKIT32/chipKIT32-MAX #575: Add "servo" wrapper class to SoftPWMServo

EmbeddedMan, 2015-06-26:
The idea here is that it will have exactly the same interface as the standard Arduino servo class, so you can just drop in SoftPWMServo.h instead of Servo.h and all existing sketches will just work.

majenkotech, 2015-06-26:
You could inherit the Servo class as a parent then and override the relevant functions.  That way the SoftPWMServo class can be passed to any functions that expect a Servo object.  Means you have to include both Servo and SoftPWMServo (if you're not on UECIDE of course)...

EmbeddedMan, 2015-06-26:
True. I've got everything done, for now, in a simple way. Next step is to
make it cleaner, as you suggest.

majenkotech, 2015-06-26:
Personally I'd like to see more interface classes defined in the core - things like the Servo interface, etc, than the libraries then use instead of defining their own interfaces.  Makes for a much cleaner system.

EmbeddedMan, 2015-06-26:
I completely agree. The trick, of course, is doing it without affecting
existing sketches.

majenkotech, 2015-06-26:
Yeah, Arduino is a real millstone round our necks...

ricklon, 2015-08-21:
Confirmed closed?

EmbeddedMan, 2015-10-25:
This should be confirmed by somebody other than me before being closed. At least test it out that it works for them.