chipKIT32/chipKIT32-MAX #588: Include all files in the variants folder for compiling.

ricklon, 2015-08-20:
Include all files in the variants folder for compiling. This the better way to do that.

```
  if (variantFolder != null)
      objectFiles.addAll(compileFiles(buildFolder, variantFolder, true,
                                      includeFolders));
```

EmbeddedMan, 2015-10-21:
Rick, where should this line go? At the very top of compileSketch() in Compiler.java?

ricklon, 2015-10-21:
I'll check. Will have feedback tonight.

--Rick

EmbeddedMan, 2015-10-22:
Rick - I think this is the last major MPIDE 'bug' to squash before it will compile the blink sketch for all boards.

ricklon, 2015-10-22:
variant folder is found in this code
```
    if (variant != null) {
            if (variant.indexOf(':') == -1) {
                Target t = Base.getTarget();
                variantPath = t.getVariantFolder(variant).getAbsolutePath();
            } else {
                Target t = Base.targetsTable.get(variant.substring(0,
variant.indexOf(':')));
                variantPath = t.getVariantFolder(variant).getAbsolutePath();
            }
        }

        logger.debug("Variant Path: " + variantPath);

```

ricklon, 2015-10-22:
It's stuffed inside this variable: includePaths
```
compileSketch(avrBasePath, buildPath, includePaths, configPreferences);
```

On Wed, Oct 21, 2015 at 10:43 PM, Rick Anderson <rick.rickanderson@gmail.com

> variant folder is found in this code
> ```
>     if (variant != null) {
>             if (variant.indexOf(':') == -1) {
>                 Target t = Base.getTarget();
>                 variantPath =
> t.getVariantFolder(variant).getAbsolutePath();
>             } else {
>                 Target t = Base.targetsTable.get(variant.substring(0,
> variant.indexOf(':')));
>                 variantPath =
> t.getVariantFolder(variant).getAbsolutePath();
>             }
>         }
>
>         logger.debug("Variant Path: " + variantPath);
>
> ```
>
>
>> Rick - I think this is the last major MPIDE 'bug' to squash before it
>> will compile the blink sketch for all boards.
>>
>>
>
>
>
> Co-founder
> Fair Use Building and Research (FUBAR) Labs
> http://fubarlabs.org

ricklon, 2015-10-22:
Then these files are searched for and added .c, .cpp, .S compiled in.
```
this.objectFiles.addAll(compileFiles(avrBasePath, buildPath, includePaths,
findFilesInPath(buildPath, "S", false),
findFilesInPath(buildPath, "c", false),
findFilesInPath(buildPath, "cpp", false),
configPreferences));
```
But it looks for "buildPath" in this portion of code.

--RIck

On Wed, Oct 21, 2015 at 10:44 PM, Rick Anderson <rick.rickanderson@gmail.com

> It's stuffed inside this variable: includePaths
> ```
> compileSketch(avrBasePath, buildPath, includePaths, configPreferences);
> ```
>
> On Wed, Oct 21, 2015 at 10:43 PM, Rick Anderson <
>
>> variant folder is found in this code
>> ```
>>     if (variant != null) {
>>             if (variant.indexOf(':') == -1) {
>>                 Target t = Base.getTarget();
>>                 variantPath =
>> t.getVariantFolder(variant).getAbsolutePath();
>>             } else {
>>                 Target t = Base.targetsTable.get(variant.substring(0,
>> variant.indexOf(':')));
>>                 variantPath =
>> t.getVariantFolder(variant).getAbsolutePath();
>>             }
>>         }
>>
>>         logger.debug("Variant Path: " + variantPath);
>>
>> ```
>>
>>
>>> Rick - I think this is the last major MPIDE 'bug' to squash before it
>>> will compile the blink sketch for all boards.
>>>
>>>
>>
>>
>>
>> Co-founder
>> Fair Use Building and Research (FUBAR) Labs
>> http://fubarlabs.org
>>
>
>
>
> Co-founder
> Fair Use Building and Research (FUBAR) Labs
> http://fubarlabs.org

EmbeddedMan, 2015-10-22:
Rick,

Right, but I don't see how to get all of the files in the variant folder to
compile. Your initial code snippit doesn't compile.

Brian

ricklon, 2015-10-22:
So it finally drills down to:
```
private List<File> compileFiles(String avrBasePath,
String buildPath,
ArrayList<String> includePaths,
List<File> sSources, List<File>
cSources,
List<File> cppSources,
HashMap<String, String> configPreferences) throws RunnerException {

List<File> objectPaths = new ArrayList<File>();

for (File file : sSources) {
String objectPath = buildPath + File.separator + file.getName()
+ ".o";
objectPaths.add(new File(objectPath));
execAsynchronously(getCommandCompilerS(avrBasePath, includePaths,
file.getAbsolutePath(), objectPath, configPreferences));
}

for (File file : cSources) {
String objectPath = buildPath + File.separator + file.getName()
+ ".o";
objectPaths.add(new File(objectPath));
execAsynchronously(getCommandCompilerC(avrBasePath, includePaths,
file.getAbsolutePath(), objectPath, configPreferences));
}

for (File file : cppSources) {
String objectPath = buildPath + File.separator + file.getName()
+ ".o";
objectPaths.add(new File(objectPath));
execAsynchronously(getCommandCompilerCPP(avrBasePath, includePaths,
file.getAbsolutePath(), objectPath, configPreferences));
}

return objectPaths;
}

```

EmbeddedMan, 2015-10-30:
I can confirm that this works as WiFire is now building without compile errors.