chipKIT32/chipKIT32-MAX #621: Fix over-ridden compiler options in boards.txt for -G1024

EmbeddedMan, 2015-10-21:
Many of the boards in boards.txt that have 64K RAM or less (which is most of our boards at this point) can benefit from the -G1024 compiler options (so says Jason, and I'd believe him), and we are not currently overriding the default compiler flags (in platforms.txt) for all of these boards.

So fix it, darn it.

EmbeddedMan, 2015-10-21:
Ended up realizing that we need to change the variant level boards.txt file to affect this change properly. Removing the top level boards.txt will prevent confusion in the future. All of the boards that need this -G1024 option also need their custom compiler flags updated with the values from platforms.txt.

ricklon, 2015-10-22:
Good idea!

EmbeddedMan, 2015-10-24:
I just realized why removing the boards.txt is not a good idea. When we get the code working to pull chipKIT-core into MPIDE when we do a build (rather than have separate core files like we do now) then this file will come back.

In any case, the original issue should now be fixed.