chipKIT32/chipKIT32-MAX #623: Remove all vestiges of plilb.h from source

EmbeddedMan, 2015-10-21:
There are a few variants that still include plib.h. Remove these includes.

EmbeddedMan, 2015-10-21:
Turns out that the quick240 board uses mSYSTEMLock() and mSYSTEMUnlock() from plib.h. 
Turns out there are actually four boards that _should_ be using these plib.h macros as they try to write to the OSCCON register to turn off the secondary oscillator.
So fixing this issue will require moving those macros (and the DmaSuspend() that they depend on) into our core libraries.

EmbeddedMan, 2015-10-21:
Turns out that there is a huge rabbit hole here. To replace the mSYSTEMUlock() macro from plib.h, we need to implement DmaSuspend(), which is part of plib.h, and which has different code for each of the three different types of DMA on PIC32. I'm not sure re-writing this section of the plib.h system is worth it at this point. It is unlikely that users will have DMA already turned on when the _board_init() function is called from Board_Data.c on boot time. So I'm just going to ignore DMA for this unlock sequence.

majenkotech, 2015-10-21:
When I finally get round to finishing my DMA abstraction layer I will add a function dma_suspend() and dma_resume() for you.

EmbeddedMan, 2015-10-21:
You are so awesome. Thanks Matt!

EmbeddedMan, 2015-10-24:
The original issue should now be fixed. No boards or libraries include plib.h anymore.