chipKIT32/chipKIT32-MAX #633: USB driver problems

nieknooijens, 2016-09-21:
Hi there

I'm trying to implement a generic USB HID device using the max32
so far when I choose "MAX32 with USB for serial" and connect an USB cable to pins 24,25,26 and GND my PC recognizes a second FTDI device with a serial port on /dev/ttyACM0
Serial.println() is printed here, so far so good!
but I want to re-purpose the USB controller. so I included <HardwareSerial_usb.h> and called some of the functions, setting a descriptor etc.

here's the problem, with "MAX32 with serial for USB" everything compiles just fine, but my USB commands are overwritten!
when compiling with the regular "MAX32" profile suddenly my linker cannot find any of the relevant functions!!
`/tmp/build6eadbbed8f11686147128e78e9a69ff7.tmp/sketch/USB.cpp.o: In function USB::USB()':
/tmp/build6eadbbed8f11686147128e78e9a69ff7.tmp/sketch/USB.cpp:19: undefined reference to usb_string_descriptor'
/tmp/build6eadbbed8f11686147128e78e9a69ff7.tmp/sketch/USB.cpp:20: undefined reference to usb_initialize'`

so... how do I fix this?

JacobChrist, 2016-09-21:
Here is HID Mouse project we created here:

https://github.com/pontech/MouseHID

This was last tested with MPIDE, don't know how it works with the latest version of chipKIT-core.

If you get it working let me know (here).

nieknooijens, 2016-09-21:
wasn't MPIDE support for the chipkit line discontinued recently?

JacobChrist, 2016-09-21:
Its discontinued in that there is no planed future releases by the people that have been maintaining it.  There are installs still available.   The MPIDE features that were important to chipKIT were rolled into the Arduino IDE.  There has been much effort to make 028 MPIDE code work in chipKIT-core but not everything has been tested.

nieknooijens, 2016-09-22:
ok, I got mplab X IDE running and then I discovered your example is a pde file (pre 1.0 arduino!) 
it seems that it cannot find the chipKITUSBDevice.h library, seems that I have to figure out how to get that.

majenkotech, 2016-09-22:
It's part of the big set of ancient libraries Digilent bundle for the
Network Shield:
https://reference.digilentinc.com/chipkit_shield_network/chipkit_shield_network

nieknooijens, 2016-09-23:
Ok I downloaded the mous demo, the last version of MPIDE, fixed all the file paths (USB is in caps and it was written in lowercase, on windows that's not a problem but in linux it is!)
I'm now stuck on this error.
[image]

WTF!??

majenkotech, 2016-09-23:
Maybe it needs to have 

    #include <stdint.h>

added to the start of it?

nieknooijens, 2016-09-26:
@majenkotech yeap that fixed a lot, still had to fix some library paths again :-1:  (linux is Case SensiTive!) I now end up with some undefined variables:
[image]

majenkotech, 2016-09-26:
I often run up against that very same case sensitivity problem with Digilent's libraries. They do seem to have a bit of a slap-dash approach to their includes.  One-liners involving `sed -i` are your friend ;)

They also seem to quite like creating their own non-standard types that mirror standard types but with a different capitalization, or lacking a _t or something similar, very nasty.

Anyway, those errors you see there are because the sketch is written for a specific board (the Quick240) and those are the symbolic names for IO pins in the "Kard" slots.  Just change the `#define` entries at the top of the sketch.

A better option, though, would be to switch to UECIDE instead for your IDE (http://uecide.org/download.php) and install the "Harmony Testing" core for chipKIT. That gives you full USB HID built in (though it is experimental) and should just work out of the box. If it doesn't I can fix it for you so it does :)

nieknooijens, 2016-09-26:
well, I installed the uecide-0.8.9-pre5_all.deb (the uecide-0.8.8alpha22 has dependency problems, I only have Java8, not 7 available) and while it works (got my hello world uploaded to the MAX32 as well as my sensor readout code) at the same time it doesn't
this happens with USB set to the default "two CDC/ACM plus HID"
keyboard example:
[image]
mouse example:
[image]

this happens with HID only and with "one CDC/ACM Plus HID":
[image]

just "two CDC/ACM" (without hid) and "one CDC/ACM" will result in undefined mouse functions (obviously)

majenkotech, 2016-09-26:
Ah yes, one small tweak you will need to make. Find the file 

    ~/.uecide/cores/chipkit-harmony/core.txt

Find the line starting:

    #core.library.harmony-usbfs=....

and remove the `#`

The problem is there's two different drivers - full speed (MX) and high speed (MZ) and no way at the moment of selecting between them. The MZ (hs) one doesn't work anyway at the moment. So just remove the # on the usbfs one to enable the fs driver. You'll need to either restart the IDE or go to "Help -> Debug -> Rebuilt internal structures" for it to pick up the change.

nieknooijens, 2016-09-26:
[image]

and still this doesn't want to go away:
[image]

majenkotech, 2016-09-26:
You seem to have uncommented them both - you only want the usbfs one, not the usbhs one.

nieknooijens, 2016-09-26:
yeah, as I already pointed out, commenting the usbhs one gives me this:
[image]

majenkotech, 2016-09-26:
Ok, I think I have found the missing `#endif`. It should be on line 600 of that file. I have a few more changes to implement as well that I'd like to get going, so I'll issue a new version of the core later today.

majenkotech, 2016-09-26:
Ok, new core available. I have updated it to be in line with chipKIT-core 1.3.1 now. I have also defaulted to the usbfs driver with a per-board option to override it when I get the MZ system working.

nieknooijens, 2016-09-26:
Yey that works! now I have to change the device descriptors and the hid classes :+1:

closed 2016-09-26 by nieknooijens

majenkotech, 2016-09-26:
What are you trying to change them to? You should be able to do everything you need to as-is. You have access to the HID object for sending raw HID requests:

    HID.sendReport(uint8_t *buffer, size_t len)

You can also receive with a callback:

    HID.onReceive(func);

`func` is defined as:

    void func(int type, uint8_t *buffer, uint32_t len)